MarginallyClever/Robot-Overlord-App #24: OpenGL exception on quit

i-make-robots, 2016-08-31:
Exception in thread "AWT-EventQueue-0" com.jogamp.opengl.GLException: Caught NativeWindowException: DC not released: GDISurface[ displayHandle 0x0
, surfaceHandle 0x52011cf4
, size 857x704
, UOB[ OWNS_SURFACE | OWNS_DEVICE | WINDOW_INVISIBLE ]
, WindowsWGLGraphicsConfiguration[DefaultGraphicsScreen[WindowsGraphicsDevice[type .windows, connection decon, unitID 0, handle 0x0, owner false, NullToolkitLock[obj 0x76933a78]], idx 0], pfdID 9, ARB-Choosen true,
	requested GLCaps[rgba 8/8/8/1, opaque, accum-rgba 0/0/0/0, dp/st/ms 16/0/4, sample-ext default, one, mono  , hw, GLProfile[GL4bc/GL4bc.hw], on-scr[.]],
	chosen    GLCaps[wgl vid 9 arb: rgba 8/8/8/8, opaque, accum-rgba 0/0/0/0, dp/st/ms 16/0/4, sample-ext default, dbl, mono  , hw, GLProfile[GL4bc/GL4bc.hw], offscr[fbo]]]
, surfaceLock <673cfebd, 25dd3b36>[count 1, qsz 0, owner <AWT-EventQueue-0>]
, GDIDummyUpstreamSurfaceHook[pixel 857x704]
, upstreamSurface false ], isWindow false, werr 0, thread: AWT-EventQueue-0 on thread AWT-EventQueue-0
	at com.jogamp.opengl.GLException.newGLException(GLException.java:76)
	at com.jogamp.opengl.awt.GLJPanel$8.run(GLJPanel.java:1502)
	at com.jogamp.opengl.Threading.invoke(Threading.java:223)
	at com.jogamp.opengl.awt.GLJPanel.dispose(GLJPanel.java:487)
	at com.jogamp.opengl.awt.GLJPanel.removeNotify(GLJPanel.java:688)
	at com.jogamp.opengl.awt.GLJPanel.destroy(GLJPanel.java:513)
	at com.jogamp.opengl.awt.GLJPanel$2.run(GLJPanel.java:312)
	at com.jogamp.nativewindow.awt.AWTWindowClosingProtocol$WindowClosingAdapter.windowClosing(AWTWindowClosingProtocol.java:71)
	at java.awt.AWTEventMulticaster.windowClosing(Unknown Source)
	at java.awt.Window.processWindowEvent(Unknown Source)
	at javax.swing.JFrame.processWindowEvent(Unknown Source)
	at java.awt.Window.processEvent(Unknown Source)
	at java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
Caused by: com.jogamp.nativewindow.NativeWindowException: DC not released: GDISurface[ displayHandle 0x0
, surfaceHandle 0x52011cf4
, size 857x704
, UOB[ OWNS_SURFACE | OWNS_DEVICE | WINDOW_INVISIBLE ]
, WindowsWGLGraphicsConfiguration[DefaultGraphicsScreen[WindowsGraphicsDevice[type .windows, connection decon, unitID 0, handle 0x0, owner false, NullToolkitLock[obj 0x76933a78]], idx 0], pfdID 9, ARB-Choosen true,
	requested GLCaps[rgba 8/8/8/1, opaque, accum-rgba 0/0/0/0, dp/st/ms 16/0/4, sample-ext default, one, mono  , hw, GLProfile[GL4bc/GL4bc.hw], on-scr[.]],
	chosen    GLCaps[wgl vid 9 arb: rgba 8/8/8/8, opaque, accum-rgba 0/0/0/0, dp/st/ms 16/0/4, sample-ext default, dbl, mono  , hw, GLProfile[GL4bc/GL4bc.hw], offscr[fbo]]]
, surfaceLock <673cfebd, 25dd3b36>[count 1, qsz 0, owner <AWT-EventQueue-0>]
, GDIDummyUpstreamSurfaceHook[pixel 857x704]
, upstreamSurface false ], isWindow false, werr 0, thread: AWT-EventQueue-0
	at jogamp.nativewindow.windows.GDISurface.unlockSurfaceImpl(GDISurface.java:121)
	at jogamp.nativewindow.ProxySurfaceImpl.unlockSurface(ProxySurfaceImpl.java:229)
	at jogamp.opengl.GLDrawableImpl.unlockSurface(GLDrawableImpl.java:332)
	at jogamp.opengl.GLContextImpl.destroy(GLContextImpl.java:509)
	at com.jogamp.opengl.awt.GLJPanel$OffscreenBackend.destroy(GLJPanel.java:1816)
	at com.jogamp.opengl.awt.GLJPanel$8.run(GLJPanel.java:1489)
	... 32 more

gouessej, 2016-09-01:
Please reproduce your bug on a tiny test case and fill a bug report against JOGL itself.

i-make-robots, 2016-09-01:
I don't know when I'll have time to do that.  One morning I woke up and the bug appeared across all of my projects that use jogamp.

gouessej, 2017-04-18:
We can't fix a bug that we can't reproduce.

i-make-robots, 2017-04-18:
Help me help you. I get this on across devices and projects using Jogamp. I'm sure it's something I'm doing. I clone, run debug, and bam!

i-make-robots, 2017-07-29:
if i comment out all calls to DrawPanel.display() then the bug goes away.  I so much as 

  GL2 gl2 = glautodrawable.getGL().getGL2();

the bug reappears.

gouessej, 2017-07-31:
We still need a more straightforward solution to reproduce this bug.

gouessej, 2017-08-01:
Do you succeed in reproducing your bug (even intermittently) with one of our tests? Maybe you can try to dispose the GLJPanel earlier when exiting.

i-make-robots, 2017-08-01:
I have not.  I've also tried removing elements from my app in the hopes
that I pear down until it stops happening.  then the intermittent nature
becomes more apparent.  Something about mutlithreading, maybe?

gouessej, 2017-08-01:
Multithreading? As long as you refrain yourself from making the OpenGL context current on several threads and if you respect Swing threading when using Swing components, it should work. I'm going to look at your code once again, maybe I missed something.

i-make-robots, 2017-08-01:
be sure to look at the dev branch.

gouessej, 2017-08-01:
By the way, replace new GLU() by GLU.createGLU(GL), create it in GLEventListener.init(GLAutoDrawable). Your current code creates a not very capable GLU instance, you might get an UnsupportedOperationException.

Don't use an FPSAnimator with V-sync (setSwapInterval), use either the former or the latter.

Don't pass indirect NIO buffers to JOGL, rather create one of multiple direct NIO buffers that you'll reuse.

Don't use EXIT_ON_CLOSE + call System.exit(), use another default close operation (hide on close or do nothing on close), override windowClosed or windowClosing. You can dispose the panel in one of those methods before exiting but after stopping the animator.

i-make-robots, 2017-08-01:
Huh.  Changing system.exit() to mainFrame.dispose() might have done the
trick.  I just ran it five times in a row without an exit exception.

i-make-robots, 2017-08-01:
> Ph: +1 (604) 259-9564 <(604)%20259-9564>
> Mo: +1 (604) 916-2281 <(604)%20916-2281>

I removed setSwapInterval.

Where do you see NIO buffers being passed to jogl?  I searched the code
base and don't see it.

i-make-robots, 2017-08-01:
I removed the setSwapInterval.  where do you see NIO buffers being passed?  My search didn't find it.

i-make-robots, 2017-08-01:
fixed in 4a7981d082e595e581258a2949b5e61b6b78bd20 ?  Will reopen if it reappears.  Thanks, @gouessej !

closed 2017-08-01 by i-make-robots

i-make-robots, 2017-08-01:
https://twitter.com/aggrav8d/status/892430674136674304

gouessej, 2017-08-01:
There is another problem with your code. You shouldn't store any GL instance into a field because it can become invalid at any time and you could be tempted to use such an object when the OpenGL context isn't current on the current thread. Rather call GLContext.getCurrentGL().

The indirect NIO buffer is still there:
https://github.com/MarginallyClever/Robot-Overlord-App/blob/dev/src/main/java/com/marginallyclever/robotOverlord/RobotOverlord.java#L562

gouessej, 2017-08-01:
By the way, I'm not on Twitter, this guy obtained my pseudonym somehow after I left this microblogging tool. Don't hesitate to bump me if there is another problem.

You should use my OBJ importer as a source of inspiration.

i-make-robots, 2017-08-01:
I didn't find you on twitter, either.  I did the best I could to give credit.  The stack thing I got from one of the jogamp examples, iirc.  I don't remember writing it.

I'm confused.  Didn't you say to create the GLU in init()?  I wouldn't need to do that unless I was storing it for later, which is why I added them.  but I shouldn't store the GL because it could become invalid.  So why would I need to create a GLU in init()?  What did I miss?

gouessej, 2017-08-02:
GLU != GL.

Yes you should create the GLU instance in GLEventListener.init(GLAutoDrawable) by calling GLU.createGLU(GL). When you're inside GLEventListener.init(GLAutoDrawable), you can obviously obtain the GL instance by calling GLAutoDrawable.getGL() which will return something not null because you're inside GLEventListener.init(GLAutoDrawable) which helps greatly to be sure to have a current OpenGL context. Moreover, when you create GLU by simply calling `new GLU()`, you create a dummy implementation that supports a very limited subset of GLU, this subset will become even smaller because it currently contains some features that will be moved into GLUgl2 or in another implementation. Therefore, you might get an UnsupportedOperationException by switching to a more recent version of JOGL or simply by calling another method only supported by GLUgl2 or GLUgl2es1. When you pass a GL instance to GLU.createGLU(GL), it's only used to determine which implementation of GLU to choose. You need to create a GLU in GLEventListener.init(GLAutoDrawable) because you need to pass a valid GL instance to GLU.createGLU(GL). If you call GLU.createGLU() outside of any GLEventListener method, you might get a GLException because there would be no current OpenGL context on the current thread.

The GL instance mustn't be used when the OpenGL context isn't current on the current thread and it can become invalid in some circumstances, for example when a panel uses a FBO under the hood (typically when using GLJPanel which is your case), when resizing a window, ...

i-make-robots, 2017-08-02:
ok, i'll keep the GLU as a class variable and create it in init().

gouessej, 2017-08-03:
I'll have to look at the source code of the GLU class (which is generated), maybe I will write a less confusing method to create such instances. Another problem with your code is that you might rely on the default profile whereas it's not the same on all OS/hardware combinations. I'll have to read some of your classes once more to confirm that.