MarginallyClever/Robot-Overlord-App #72: Serial communication error

joladouille, 2020-08-25:
i have plug and flash a Arduino mega 2560 to my computer and when i run robot overlord and connect the sixi 2 robot with the Arduino board i have big red message in the eclipse console

for the connection i made : 
[image]
and
[image]

after this i have this message on the console : 
```
Caused by: com.jogamp.opengl.GLException: Caught NumberFormatException: For input string: "2,095" on thread AWT-EventQueue-0
	at com.jogamp.opengl.GLException.newGLException(GLException.java:76)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1311)
	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1131)
	at com.jogamp.opengl.awt.GLJPanel$OffscreenBackend.doPaintComponent(GLJPanel.java:2094)
	at com.jogamp.opengl.awt.GLJPanel.paintComponent(GLJPanel.java:568)
	at javax.swing.JComponent.paint(Unknown Source)
	at javax.swing.JComponent.paintToOffscreen(Unknown Source)
	at javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(Unknown Source)
	at javax.swing.RepaintManager$PaintManager.paint(Unknown Source)
	at javax.swing.RepaintManager.paint(Unknown Source)
	at javax.swing.JComponent._paintImmediately(Unknown Source)
	at javax.swing.JComponent.paintImmediately(Unknown Source)
	at javax.swing.RepaintManager$4.run(Unknown Source)
	at javax.swing.RepaintManager$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at javax.swing.RepaintManager.paintDirtyRegions(Unknown Source)
	at javax.swing.RepaintManager.paintDirtyRegions(Unknown Source)
	at com.jogamp.opengl.util.AWTAnimatorImpl$1.run(AWTAnimatorImpl.java:164)
	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
Caused by: java.lang.NumberFormatException: For input string: "2,095"
	at sun.misc.FloatingDecimal.readJavaFormatString(Unknown Source)
	at sun.misc.FloatingDecimal.parseDouble(Unknown Source)
	at java.lang.Double.parseDouble(Unknown Source)
	at com.marginallyclever.robotOverlord.entity.scene.dhRobotEntity.sixi2.Sixi2Model.sendCommand(Sixi2Model.java:252)
	at com.marginallyclever.robotOverlord.entity.scene.dhRobotEntity.sixi2.Sixi2Live.sendCommand(Sixi2Live.java:183)
	at com.marginallyclever.robotOverlord.entity.scene.dhRobotEntity.sixi2.Sixi2.update(Sixi2.java:116)
	at com.marginallyclever.robotOverlord.entity.Entity.update(Entity.java:63)
	at com.marginallyclever.robotOverlord.entity.Entity.update(Entity.java:63)
	at com.marginallyclever.robotOverlord.RobotOverlord.display(RobotOverlord.java:788)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:691)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:673)
	at com.jogamp.opengl.awt.GLJPanel$Updater.display(GLJPanel.java:1445)
	at com.jogamp.opengl.awt.GLJPanel$10.run(GLJPanel.java:1520)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1277)
	... 31 more
java.lang.reflect.InvocationTargetException
	at java.awt.EventQueue.invokeAndWait(Unknown Source)
	at java.awt.EventQueue.invokeAndWait(Unknown Source)
	at javax.swing.SwingUtilities.invokeAndWait(Unknown Source)
	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:92)
	at com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:451)
	at com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:178)
	at java.util.TimerThread.mainLoop(Unknown Source)
	at java.util.TimerThread.run(Unknown Source)
Caused by: com.jogamp.opengl.GLException: Caught NumberFormatException: For input string: "2,095" on thread AWT-EventQueue-0
	at com.jogamp.opengl.GLException.newGLException(GLException.java:76)
[... 1325 lines cut ...]
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
Caused by: java.lang.NumberFormatException: For input string: "2,095"
	at sun.misc.FloatingDecimal.readJavaFormatString(Unknown Source)
	at sun.misc.FloatingDecimal.parseDouble(Unknown Source)
	at java.lang.Double.parseDouble(Unknown Source)
	at com.marginallyclever.robotOverlord.entity.scene.dhRobotEntity.sixi2.Sixi2Model.sendCommand(Sixi2Model.java:252)
	at com.marginallyclever.robotOverlord.entity.scene.dhRobotEntity.sixi2.Sixi2Live.sendCommand(Sixi2Live.java:183)
	at com.marginallyclever.robotOverlord.entity.scene.dhRobotEntity.sixi2.Sixi2.update(Sixi2.java:116)
	at com.marginallyclever.robotOverlord.entity.Entity.update(Entity.java:63)
	at com.marginallyclever.robotOverlord.entity.Entity.update(Entity.java:63)
	at com.marginallyclever.robotOverlord.RobotOverlord.display(RobotOverlord.java:788)
	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:691)
	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:673)
	at com.jogamp.opengl.awt.GLJPanel$Updater.display(GLJPanel.java:1445)
	at com.jogamp.opengl.awt.GLJPanel$10.run(GLJPanel.java:1520)
	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1277)
	... 31 more
```
*i don't known if it is the full message because the console blink forever*

the 2 Serial led of the arduino *(TX and RX)* blinking forever as well.

i-make-robots, 2020-08-25:
I thought I dealt with this already. 
European number formats! Grr!

> On Aug 25, 2020, at 2:22 AM, joladouille <notifications@github.com> wrote:
> 
> ﻿
> i have plug and flash a Arduino mega 2560 to my computer and when i run robot overlord and connect the sixi 2 robot with the Arduino board i have big red message in the eclipse console
> 
> for the connection i made :
> 
> and
> 
> 
> after this i have this message on the console :
> 
> Caused by: com.jogamp.opengl.GLException: Caught NumberFormatException: For input string: "2,095" on thread AWT-EventQueue-0
> 	at com.jogamp.opengl.GLException.newGLException(GLException.java:76)
> 	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1311)
> 	at jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1131)
> 	at com.jogamp.opengl.awt.GLJPanel$OffscreenBackend.doPaintComponent(GLJPanel.java:2094)
> 	at com.jogamp.opengl.awt.GLJPanel.paintComponent(GLJPanel.java:568)
> 	at javax.swing.JComponent.paint(Unknown Source)
> 	at javax.swing.JComponent.paintToOffscreen(Unknown Source)
> 	at javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(Unknown Source)
> 	at javax.swing.RepaintManager$PaintManager.paint(Unknown Source)
> 	at javax.swing.RepaintManager.paint(Unknown Source)
> 	at javax.swing.JComponent._paintImmediately(Unknown Source)
> 	at javax.swing.JComponent.paintImmediately(Unknown Source)
> 	at javax.swing.RepaintManager$4.run(Unknown Source)
> 	at javax.swing.RepaintManager$4.run(Unknown Source)
> 	at java.security.AccessController.doPrivileged(Native Method)
> 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
> 	at javax.swing.RepaintManager.paintDirtyRegions(Unknown Source)
> 	at javax.swing.RepaintManager.paintDirtyRegions(Unknown Source)
> 	at com.jogamp.opengl.util.AWTAnimatorImpl$1.run(AWTAnimatorImpl.java:164)
> 	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
> 	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
> 	at java.awt.EventQueue.access$500(Unknown Source)
> 	at java.awt.EventQueue$3.run(Unknown Source)
> 	at java.awt.EventQueue$3.run(Unknown Source)
> 	at java.security.AccessController.doPrivileged(Native Method)
> 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
> 	at java.awt.EventQueue.dispatchEvent(Unknown Source)
> 	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
> 	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
> 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
> 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
> 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
> 	at java.awt.EventDispatchThread.run(Unknown Source)
> Caused by: java.lang.NumberFormatException: For input string: "2,095"
> 	at sun.misc.FloatingDecimal.readJavaFormatString(Unknown Source)
> 	at sun.misc.FloatingDecimal.parseDouble(Unknown Source)
> 	at java.lang.Double.parseDouble(Unknown Source)
> 	at com.marginallyclever.robotOverlord.entity.scene.dhRobotEntity.sixi2.Sixi2Model.sendCommand(Sixi2Model.java:252)
> 	at com.marginallyclever.robotOverlord.entity.scene.dhRobotEntity.sixi2.Sixi2Live.sendCommand(Sixi2Live.java:183)
> 	at com.marginallyclever.robotOverlord.entity.scene.dhRobotEntity.sixi2.Sixi2.update(Sixi2.java:116)
> 	at com.marginallyclever.robotOverlord.entity.Entity.update(Entity.java:63)
> 	at com.marginallyclever.robotOverlord.entity.Entity.update(Entity.java:63)
> 	at com.marginallyclever.robotOverlord.RobotOverlord.display(RobotOverlord.java:788)
> 	at jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:691)
> 	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:673)
> 	at com.jogamp.opengl.awt.GLJPanel$Updater.display(GLJPanel.java:1445)
> 	at com.jogamp.opengl.awt.GLJPanel$10.run(GLJPanel.java:1520)
> 	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1277)
> 	... 31 more
> java.lang.reflect.InvocationTargetException
> 	at java.awt.EventQueue.invokeAndWait(Unknown Source)
> 	at java.awt.EventQueue.invokeAndWait(Unknown Source)
> 	at javax.swing.SwingUtilities.invokeAndWait(Unknown Source)
> 	at com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:92)
[... 859 lines cut ...]
> 	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
> 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
> 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
> 	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
> 	at java.awt.EventDispatchThread.run(Unknown Source)
> Caused by: java.lang.NumberFormatException: For input string: "2,095"
> 	at sun.misc.FloatingDecimal.readJavaFormatString(Unknown Source)
> 	at sun.misc.FloatingDecimal.parseDouble(Unknown Source)
> 	at java.lang.Double.parseDouble(Unknown Source)
> 	at com.marginallyclever.robotOverlord.entity.scene.dhRobotEntity.sixi2.Sixi2Model.sendCommand(Sixi2Model.java:252)
> 	at com.marginallyclever.robotOverlord.entity.scene.dhRobotEntity.sixi2.Sixi2Live.sendCommand(Sixi2Live.java:183)
> 	at com.marginallyclever.robotOverlord.entity.scene.dhRobotEntity.sixi2.Sixi2.update(Sixi2.java:116)
> 	at com.marginallyclever.robotOverlord.entity.Entity.update(Entity.java:63)
> 	at com.marginallyclever.robotOverlord.entity.Entity.update(Entity.java:63)
> 	at com.marginallyclever.robotOverlord.RobotOverlord.display(RobotOverlord.java:788)
[... cut ...]
> 	at jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:673)
> 	at com.jogamp.opengl.awt.GLJPanel$Updater.display(GLJPanel.java:1445)
> 	at com.jogamp.opengl.awt.GLJPanel$10.run(GLJPanel.java:1520)
> 	at jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1277)
> 	... 31 more
> java.lang.reflect.InvocationTargetException
> 	at java.awt.EventQueue.invokeAndWait(Unknown Source)
> 	at java.awt.EventQueue.invokeAndWait(Unknown Source)
> 	at javax.swing.SwingUtilities.invokeAndWait(Unknown So

i-make-robots, 2020-08-25:
> Caused by: java.lang.NumberFormatException: For input string: "2,095"
> 	at sun.misc.FloatingDecimal.readJavaFormatString(Unknown Source)
> 	at sun.misc.FloatingDecimal.parseDouble(Unknown Source)
> 	at [...]

Yeah, it's reading serial messages (always US format numbers) using your locale (which might be non-US) and breaking when it gets to the wrong format.  Fix is in, please try the nightly build that will be generated in a few minutes.

i-make-robots, 2020-08-25:
Some JUnit tests in CI were failing.  It has been fixed and the nightly build is here: https://github.com/MarginallyClever/Robot-Overlord-App/releases/download/Nightly/robot-overlord-nightly-20200825-302c4b.jar
Please let me know if this fixes the issue.

joladouille, 2020-11-03:
salut et désoler pour le long long temps de réponse
tout les problème on l'aire d'être corrigé merci pour le travail et le temps passer a m'aider a résoudre le problème

closed 2020-11-03 by joladouille