MarginallyClever/Robot-Overlord-App #74: how add a custom 6 axis robot (and step file)

joladouille, 2020-11-07:
hello, I can't find the option in this video to add your own robot arm :
https://www.youtube.com/watch?v=pd4bLEBBxGM

I'd like to add this robot I'm building to control it with your firmware/software :
https://www.thingiverse.com/thing:3327968/files

the person who made this arm has made a step file assembly
Would it be possible to add this file type directly?

i-make-robots, 2020-11-07:
In src/main/resources/META-INF/services/ is a file "com.marginallyclever.robotOverlord.entity.Entity" which lists Entities that can be added to the Scene.  line 5 is commented out:
````# DHRobots
#com.marginallyclever.robotOverlord.entity.scene.robotEntity.DHBuilderApp````

Remove the # on the start and it will now appear in the app.  I haven't used it in a while, completely focused on making the Sixi robot easier to control.

i-make-robots, 2020-11-07:
Also that should be "com.marginallyclever.robotOverlord.entity.scene.dhRobotEntity.DHBuilderApp" in the latest edition.  the name must match the fully qualified class name.

joladouille, 2020-11-08:
I have uncomment the line 5 in com.marginallyclever.robotOverlord.entity.Entity and i have this
 error
```java
Exception in thread "AWT-EventQueue-0" java.util.ServiceConfigurationError: com.marginallyclever.robotOverlord.entity.Entity: Provider com.marginallyclever.robotOverlord.entity.scene.robotEntity.DHBuilderApp not found
	at java.util.ServiceLoader.fail(ServiceLoader.java:239)
	at java.util.ServiceLoader.access$300(ServiceLoader.java:185)
	at java.util.ServiceLoader$LazyIterator.nextService(ServiceLoader.java:372)
	at java.util.ServiceLoader$LazyIterator.next(ServiceLoader.java:404)
	at java.util.ServiceLoader$1.next(ServiceLoader.java:480)
	at com.marginallyclever.robotOverlord.swingInterface.commands.CommandAddEntity.actionPerformed(CommandAddEntity.java:52)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:289)
	at java.awt.Component.processMouseEvent(Component.java:6539)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6304)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
	at java.awt.Container.dispatchEventImpl(Container.java:2283)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```
[image]
i have this 3 error detect by eclipse

i-make-robots, 2020-11-08:
is this the dev branch?

i-make-robots, 2020-11-09:
Any luck?

joladouille, 2020-11-09:
>> is this the dev branch?
maybe yes
i try in master tomorrow

i-make-robots, 2021-12-13:
Hello!  I'm wondering if this arm is now in the system in your local branch.  Can I help in any way?

i-make-robots, 2021-12-16:
[image]

i-make-robots, 2021-12-16:
Unfortunately these files are huge and Fusion is not exporting them in the new position. very annoying.

i-make-robots, 2021-12-16:
[image]
They're much too highly detailed.  I need these decimated down to 1-2mb each.

closed 2022-08-06 by github-actions[bot]