MarginallyClever/Robot-Overlord-App #88: Arduino sketch files?

Drumsticksplinter, 2022-01-02:
Hi, I’m developing my own robot arm design to try and get students interested in robotics and for also my own interest. 

The robot overlord software looks really great and I’d like to try and use this to control the arm I’m building.

Are there any examples of the app communicating with say an arduino mega that would help to get the control side working? Would the app work with something like grbl mega 5x?

Thanks

i-make-robots, 2022-01-02:
This sounds like a great new Friday Facts post.  I'll make it happen.  Stay tuned to marginallyclever.com for this friday's post!  It will be a lot like [Friday Facts 4](https://www.marginallyclever.com/2021/10/friday-facts-4-how-to-marlin-polargraph/), but for robot arms like the Sixi 3.

Short answer: If your hardware can run Marlin then you can make any stepper-motor-based robot work with Robot Overlord.
[Please stay in touch](https://discord.gg/p2NRZFxXbh), I want to collaborate with teachers like you to mentor more students and make robots happen faster.

At this time GRBL is not a priority because we have an equally good system that works.  Going from 1 system to 2 systems is harder than from 2 to many.

Drumsticksplinter, 2022-01-02:
Hi Dan,

Thank you so much for your quick response. I will definitely look out for the Friday facts write up. 

Excellent, I’ll look to source a Rumba board in the near future, I’ve just got a standard ramps 1.4 to test with currently. 

With the Robot overlord app, is it possible to trigger a sequence from an input on the marlin board or separate board / plc? That feature would be amazing! 

Thanks,

Adam

i-make-robots, 2022-01-02:
At the moment there's a "step" button to walk through them.  We have a "play" button in Makelangelo software and I have yet to port it over.  Shouldn't be much trouble.  I can make that happen by friday and include it in the post.

i-make-robots, 2022-01-02:
added rewind/play/pause/step buttons to robot arm interface in cc13cace

i-make-robots, 2022-01-02:
only plays simulation when robot is connected.  same as Makelangelo software.

Drumsticksplinter, 2022-01-02:
Oh Wow! that was super quick!

Thank you.

Can these be triggered by a hardware input in some way at all?

(Issue #88)

or Android

i-make-robots, 2022-01-02:
What do you have in mind?

Drumsticksplinter, 2022-01-02:
Well I’m looking to have a robot arm that can act upon some logic, maybe sorting coloured objects, another microcontroller can take care of this, then pass a signal to marlin to play a sequence from gcode.

i-make-robots, 2022-01-03:
well if two microcontrollers can talk to each other with gcode then i don't
see a problem.  you don't even need robot overlord for that.

closed 2022-07-28 by i-make-robots