MarginallyClever/Robot-Overlord-App #89: File not found. Error

hglmz, 2022-07-27:
Hello, I was not able to run the software.
when i run star windows.bat to run the software i get the following error.

```
64 bit Windows
ERROR: The system was unable to find the specified registry key or value.
64 bit Java installed.
The current Java runtime is 1.8
finding the latest Robot-Overlord JAR...
File not found.
Press any key to continue . . .
```

When I start to run the RobotOverlord-1.7.1-with-dependencies.jar file, I get an error like the one below.

[image]

i-make-robots, 2022-07-27:
Does it work with the nightly build?  https://github.com/MarginallyClever/Robot-Overlord-App/releases/download/Nightly/robot-overlord-nightly-20220627-987cee.jar

i-make-robots, 2022-07-27:
Very likely you have more than one version of Java installed and it is getting confused.

closed 2022-07-27 by i-make-robots

hglmz, 2022-07-27:
yes, the problem was caused by this, I deleted all java sdks and installed the latest data generation. problem solved. novice error :) . thanks for the quick reply. I'm excited to build the arm.

i-make-robots, 2022-07-27:
which arm?  I have several.

hglmz, 2022-07-27:
I'm aiming to make the sixi 2, which I shot the dioramas with a miniature camera. I'm going to need a slightly longer arm for this. The camera is extremely light. I rented a kuka but it's too big for my needs.

i-make-robots, 2022-07-28:
I’m tempted to sell you one of the units I have here at cost. I don’t need two.