MarginallyClever/Robot-Overlord-App #126: Bug: File picker get wrong path on double click

Jackymancs4, 2023-05-21:
**Describe the bug**
As per discussion via Discord, I have managed to consistently replicate this.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to one any file selector, and open the window
2. Double click on any file
3. Click on "Select"
4. See that the file path have the pattern "C:\Users\{user}\RobotOverlord\Scene\{file_name}\{file_name}". Note the repeated file name.
5. Reopen the file selector.
6. See that the list is empty, while the folder does contain some files.

**Expected behavior**
The file path should point to the selected file.

**Platform (please complete the following information):**
 - Device: PC
 - OS: Windows 10
 - App Version 2.6.0 (built on ef8fb0c1b4d836e6d51e64e61d8db2f63c27b40d)

closed 2023-05-21 by i-make-robots

Jackymancs4, 2023-05-22:
I confirm the issue is fixed!

i-make-robots, 2023-05-22:
on double click the system that "jails" users to the scene path would try to open the file like a directory, cause and exception, and put the "file selection" dialog in a wierd state that produced the funky file/file name.

commit 8f0def84 (2023-05-21): fix #126 and stop deleting scene path contents
commit f309e11a (2023-05-21): Merge pull request #131 from MarginallyClever/fix-126

fix #126