MarginallyClever/Robot-Overlord-App #127: FR: Message on why the Contron

Jackymancs4, 2023-05-21:
**Is your feature request related to a problem? Please describe.**
On any newly created robot arm, clicking on the "Control panel" does nothing.

**Describe the solution you'd like**
I would like a message box telling me what is preventing the window do be opened.

**Describe alternatives you've considered**
As an alternative, I would like the button to be disabled if the conditions for the window to be opened are not met.

**Additional context**
If I create a Thor robot arm and begin to delete joints until only one remains the control panel still spawns, but with any 1-joint robot arm I created the button does nothing.

i-make-robots, 2023-05-21:
To reproduce:
- start fresh
- add entity
- add robotComponent to entity
- click "control panel" in robotComponent.

Exception message:
```
Exception in thread "AWT-EventQueue-0" java.lang.ArrayIndexOutOfBoundsException: Index 0 out of bounds for length 0
	at com.marginallyclever.robotoverlord/com.marginallyclever.robotoverlord.systems.robot.robotarm.robotpanel.jogpanel.AngleDrivePanel.<init>(AngleDrivePanel.java:33)
	at com.marginallyclever.robotoverlord/com.marginallyclever.robotoverlord.systems.robot.robotarm.robotpanel.jogpanel.JogPanel.<init>(JogPanel.java:45)
	at com.marginallyclever.robotoverlord/com.marginallyclever.robotoverlord.systems.robot.robotarm.robotpanel.RobotPanel.<init>(RobotPanel.java:42)
	at com.marginallyclever.robotoverlord/com.marginallyclever.robotoverlord.systems.robot.robotarm.ArmRobotSystem.showControlPanel(ArmRobotSystem.java:75)
	at com.marginallyclever.robotoverlord/com.marginallyclever.robotoverlord.systems.robot.robotarm.ArmRobotSystem.lambda$decorateRobot$1(ArmRobotSystem.java:64)
	at com.marginallyclever.robotoverlord/com.marginallyclever.robotoverlord.swinginterface.componentmanagerpanel.ViewElementButton.fireActionEvent(ViewElementButton.java:31)
	at com.marginallyclever.robotoverlord/com.marginallyclever.robotoverlord.swinginterface.componentmanagerpanel.ViewElementButton.lambda$new$0(ViewElementButton.java:41)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
	at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6614)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342)
	at java.desktop/java.awt.Component.processEvent(Component.java:6379)
	at java.desktop/java.awt.Container.processEvent(Container.java:2263)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:4990)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4822)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4919)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4548)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4489)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2769)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4822)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:391)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:391)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
```

i-make-robots, 2023-05-21:
[image]

i-make-robots, 2023-05-22:
fixed in 0f7e9081

Jackymancs4, 2023-05-22:
Can you document when a robot is "Initialized"?

i-make-robots, 2023-05-22:
what do you mean by "document"?  I can calculate when a robot has more than just a RobotComponent.

Jackymancs4, 2023-05-23:
I believe the official documentation lives here: https://github.com/MarginallyClever/Robot-Overlord-App/wiki

Following (more or less) https://github.com/MarginallyClever/Robot-Overlord-App/wiki/How-to-add-a-new-robot-arm , a user ends up with a non initialized robot.

When the prompt ask me "Is robot initialized?", my inner answer is "i don't know, but I guess not".

What more than a RobotComponent is needed for the most basic robot to be initialized?
Maybe a minimal troubleshoot checklist like:
- have set all your meshes?
- Are there at least 1 joint?
- Have you set D and R on those?
- Have you set the target?
- Ecc..

I suggest adding a paragraph in the docs about a minimal setup.
I can help with that, but not before I figure out how to get my 1-joint arm to be controlled.

i-make-robots, 2023-05-23:
Imho a video about how to use the tool would work better than a built-in wizard of some kind.  could be linked at the bottom of the edit arm panel, directly opening the youtube page.

Jackymancs4, 2023-05-23:
That would be soo cool!

i-make-robots, 2023-05-28:
Part of the "things blow up" has to do with mesh origins.  A big part of me wants to improve that before making a video.

i-make-robots, 2023-06-02:
This is a closely related to #148

i-make-robots, 2023-06-02:
[image]

i-make-robots, 2023-06-02:
[image]