MarginallyClever/Robot-Overlord-App #128: Allow scaling pose components

Jackymancs4, 2023-05-21:
**Is your feature request related to a problem? Please describe.**
All my models are exported in mm and when imported the get huge

**Describe the solution you'd like**
Restore the scale parameters (X,Y,Z) on the pose component. I'm pretty sure it was there on 2.5.8.

**Describe alternatives you've considered**
Allow the pose component to specify the Unit of measurement.

**Additional context**
I know that the wiki (https://github.com/MarginallyClever/Robot-Overlord-App/wiki/How-to-add-a-new-robot-arm#notes) says to export everything in cm, so this feature request is more of a Quality of Life improvement.

i-make-robots, 2023-05-22:
This one is complicated.  the best solution is to scale the model in Blender and then open it in app.  adding scale to PoseComponent causes problems with translate and rotate tool.  Adding an extra scale to mesh is also problematic.

i-make-robots, 2023-05-22:
and thank you for these great issues!  well reported and in the right place!  huzzah!

closed 2023-07-30 by github-actions[bot]