MarginallyClever/Robot-Overlord-App #129: Improve "Edit arm" layout

Jackymancs4, 2023-05-21:
**Is your feature request related to a problem? Please describe.**
The Edit arm window is quite responsive, but I don't think it should be.
By default, it is very difficult to see the various file paths.

**Describe the solution you'd like**
My suggestions:
- Set minimun width and height for the windows to something sensible like 800*600
- Keep the row height fixed. Also keep the column width fixed, exept for "Mesh".
- Align and pad "# Joint", "Texture" and "File"
- Stretch "Texture" and "File" values

**Additional context**
Screen for reference:
[image]
[image]
ture request here.

i-make-robots, 2023-05-21:
Agreed.  I used a `GridLayout` for the table and it... doesn't scale well.  it doesn't allow column sizes, either.

i-make-robots, 2023-05-21:
[image]
like this?

i-make-robots, 2023-05-21:
[image]
[image]
[image]
[image]
doesn't respect my minimum size settings....

closed 2023-05-21 by i-make-robots

Jackymancs4, 2023-05-22:
Great improvement on the layout!

Only suggestion is to slighly increase the minum column size, so that the default values (with decimals) are still readable in "cramped" mode.

[image]

i-make-robots, 2023-05-22:
https://github.com/MarginallyClever/Robot-Overlord-App/commit/a7129e781ee9fd8b79a813935dfbe403a7a8f786#diff-1201499337140b977a2079d8e7197b647ed24756fb4d103f6679ec4d185044d0R36
You can see the ground work laid out here.  I don't know off the top of my head how to make it perfect. 
I welcome a PR to make this more perfect look & feel for you.

Jackymancs4, 2023-05-22:
Yes, I reviewed those changes this morning. Yesterday I bet you would use a GridBagLayout!

Personally, I think setMinimumSize is just bugged to hell (https://stackoverflow.com/questions/10157954/java-swing-setmaximumsize-not-working & https://stackoverflow.com/questions/4061010/setmaximumsize-not-working-in-java)

>  I welcome a PR to make this more perfect look & feel for you.

I most likely will.