MarginallyClever/Robot-Overlord-App #130: Opening Edit arm automatically moves arm components

Jackymancs4, 2023-05-21:
**Describe the bug**
This one may be my fault for not understanding the expected behavior.
When I create a robot Arm, opening the "Edit arm" panel for the first time jump start computations on unknown (to me) nature, that move the arm in a final resting position.

It is impossibile to restore the previous values as "Undo" does not work.

**To Reproduce**
Steps to reproduce the behavior:
1. Create new scene
2. Add Thor arm
3. Open the "Edit Arm"
4. See the arm start moving by itself

**Expected behavior**
Opening "Edit Arm" should allow me to edit the starting values of the arm.
The current "calculated" values should be editable in the "Control panel"

**Platform (please complete the following information):**
 - Device: PC
 - OS: Windows 10
 - App Version 2.6.0

**Additional context**
This behavior also happens with newly created arms.

i-make-robots, 2023-05-21:
Thor and some of the downloadable arms do not have a 'target' entity.  when target is added the arm immediately tries to move to it.  but it is in the wrong place!

closed 2023-05-21 by i-make-robots

i-make-robots, 2023-05-23:
so... is this considered fixed?

Jackymancs4, 2023-05-23:
Yes. The PowerSwitchComponent and/or "Save" and"Cancel" in the arm editor ideas should be new feature request to be evaluated on their own.

commit e191be83 (2023-05-21): fix #130 , put target at end effector when target is created.