MarginallyClever/Robot-Overlord-App #134: Allow deferred evaluation of parameters during arm editing.

Jackymancs4, 2023-05-23:
**Is your feature request related to a problem? Please describe.**
As per discussion on Discord. Editing any value in the arm editor cause it to be instantly evaluated. This can result in an unexpected behavior (like the arm exploding) which is difficult to undo.

**Describe the solution you'd like**
"Save" or "Cancel" in the arm editor. Evaluate the parameters on on save. Please allow Reset or Undo.

**Describe alternatives you've considered**
robotComponent needs an "on/off" switch. now do i make PowerSwitchComponent or do i make it a part of RobotComponent...

**Additional context**
The body of the issue should be edited to include proper details.

i-make-robots, 2023-05-30:
If instant evaluation doesn't happen then how do you know what you changed and if the change is correct?
I think making the undo/redo work with the edit panel is the better way to go.

i-make-robots, 2023-05-30:
in branch motor-1, I set the "all meshes at root origin" selected from the start.  
most parameters now seem undo-able.
Also split "show DH" to be per-joint.

i-make-robots, 2023-05-31:
https://github.com/MarginallyClever/Robot-Overlord-App/tree/motor-1
much improved support for undo/redo.

i-make-robots, 2023-05-31:
Merged into main.  @Jackymancs4 please let me know if this meets your needs.
also updated instructions for how to build + publish an arm.  https://github.com/MarginallyClever/Robot-Overlord-App/wiki/How-to-add-a-new-robot-arm