MarginallyClever/Robot-Overlord-App #147: Edit Arm should create a separate entity for the base.

Jackymancs4, 2023-06-01:
**Is your feature request related to a problem? Please describe.**
When I create a new robot arm and set the Mesh file, the relative MeshFromFileComponent is added to the root entity. This way I cannot move it when the mesh is not correctly centered.

**Describe the solution you'd like**
Inside the root entity, I would like a dedicated Entity with the mesh of the base, so that I can position it where I want relative to the root position. This is how the Thor arm is configured.

**Describe alternatives you've considered**
I've manually created the entity and added the mesh component.

i-make-robots, 2023-06-02:
This would be more consistent.  The only hiccup is updating older arms to match.

i-make-robots, 2023-06-02:
the benefit of having the mesh in the base is... when mesh is in base, clicking the mesh selects the entire arm.  otherwise you only get the one piece.  it would be great if the meshes were NOT separate entities and then I would... get what i expect more when i click.  what i mean is.. i'm almost never clicking the mesh because I want the mesh - I want the associated joint.

closed 2023-08-09 by github-actions[bot]