MarginallyClever/Robot-Overlord-App #158: Add  a "Project Schema Version" to save files

i-make-robots, 2023-06-03:
**Is your feature request related to a problem? Please describe.**
Old files loaded in new app, breaks.
With schema version an upgrade path could be calculated.

**Describe the solution you'd like**
when Project is saved it should include a Int schemaVersion.  initial value should be 1.  
If there is no schemaVersion found on load assume 0.

**Describe alternatives you've considered**
None.

**Additional context**
How will the upgrade path be maintained?  Should all components have a version?  UGH.

Jackymancs4, 2023-06-03:
> Should all components have a version?

No, I think that a component should always load what it can, ignore the unknow and default the missing.

Structural changes like migrate RobotComponent to ArmRobotComponent if RobotComponent has a child with ArmEffectorComponent or adding the targer component when missing should be handled by migrations.

i-make-robots, 2023-06-03:
what does that look like - some kind of `MigrationEngine`?

closed 2023-06-12 by i-make-robots