MarginallyClever/Robot-Overlord-App #159: Support Wheeled Robot

Jackymancs4, 2023-06-04:
**TODO**

- [ ] Complete https://github.com/MarginallyClever/Robot-Overlord-App/pull/157 (or similar effort)
- [ ] Create WheelComponent
	- [ ] Define how to integratate it with the MotorComponent
- [ ] Discuss further

Jackymancs4, 2023-06-04:
Proposed by @i-make-robots 

```java
/**
 * A {@link WheelComponent} is used by a CarSystem to calculate the correct velocities for adjacent
 * {@link MotorComponent}s.
 *
 * @since 2.6.3
 * @author Dan Royer
 */
@ComponentDependency(components ={PoseComponent.class,MotorComponent.class})
public class WheelComponent extends Component {
    public static final String [] names = {"Normal","Omni","Mecanum"};
    public static final int TYPE_NORMAL = 0;
    public static final int TYPE_OMNI = 1;
    public static final int TYPE_MECANUM = 2;

    public final IntParameter type = new IntParameter("Type", 0);
    public final DoubleParameter diameter = new DoubleParameter("Diameter", 1);
    public final DoubleParameter width = new DoubleParameter("Width", 1);

    @Override
    public JSONObject toJSON(SerializationContext context) {
        JSONObject jo = super.toJSON(context);
        jo.put("type", type.toJSON(context));
        jo.put("diameter", diameter.toJSON(context));
        jo.put("width", width.toJSON(context));
        return jo;
    }

    @Override
    public void parseJSON(JSONObject jo, SerializationContext context) throws JSONException {
        super.parseJSON(jo, context);
        type.parseJSON(jo.getJSONObject("type"), context);
        diameter.parseJSON(jo.getJSONObject("diameter"), context);
        width.parseJSON(jo.getJSONObject("width"), context);
    }
}
```

Wanna get fancy and add friction?
What's the difference between Omni e Mecanum?

i-make-robots, 2023-06-04:
omni: https://en.wikipedia.org/wiki/Omni_wheel
mecanum: https://en.wikipedia.org/wiki/Mecanum_wheel

combined forces from all wheels have different effects.

i-make-robots, 2023-06-04:
imho there is no "integration with motor component".  it's the CarSystem that does all the work of putting these things together.

Jackymancs4, 2023-06-04:
Quoting https://en.wikipedia.org/wiki/Omni_wheel:
> Omni wheels or poly wheels, similar to [Mecanum wheels](https://en.wikipedia.org/wiki/Mecanum_wheel), are [wheels](https://en.wikipedia.org/wiki/Wheel) with small discs (called rollers) ...

> combined forces from all wheels have different effects.

Is it worth to have a distinction?

>  Define how to integrate it with the MotorComponent

i mean, "integration with motor component" through the CarSystem.
What work should it do?

I think, speaking strictly about wheels, we should look at the GripperSystem. A WheelSystem able to couple one or multiple WheelComponent to a single MotorComponent and an optional steering MotorComponent or DHComponents.

Then a WheeledRobotSystem can couple multiple Wheel+motor blocks, and do the job, of rotating Steering motorComponents and setting speeds for the driving motorComponents.

i-make-robots, 2023-06-04:
> Is it worth to have a distinction?

I believe yes. 

```java
/**
 * {@link CarComponent} references a list of {@link WheelComponent}s.  A CarSystem then uses these to move
 * the {@link com.marginallyclever.robotoverlord.entity.Entity} that owns the CarComponent.
 */
public class CarComponent extends Component {
    public final List<ReferenceParameter> wheels = new ArrayList<>();

    public CarComponent() {
        super();
    }

    @Override
    public JSONObject toJSON(SerializationContext context) {
        JSONObject json = super.toJSON(context);
        json.put("numWheels", wheels.size());
        for(int i=0;i<wheels.size();++i) {
            json.put("wheel"+i, wheels.get(i).toJSON(context));
        }
        return json;
    }

    @Override
    public void parseJSON(JSONObject jo, SerializationContext context) throws JSONException {
        super.parseJSON(jo, context);
        int numWheels = jo.getInt("numWheels");
        for(int i=0;i<numWheels;++i) {
            wheels.add(new ReferenceParameter("wheel"+i));
            wheels.get(i).parseJSON(jo.getJSONObject("wheel"+i), context);
        }
    }
}
```

i-make-robots, 2023-06-04:
start of a rough draft.  note unit tests.

https://github.com/MarginallyClever/Robot-Overlord-App/tree/car-and-wheel

i-make-robots, 2023-06-12:
done in #160

closed 2023-06-12 by i-make-robots