MarginallyClever/Robot-Overlord-App #161: Feedback on Wheeled Robots

Jackymancs4, 2023-06-12:
1. I think Car (as in CarComponent) shound be renamed as it referes to a specific kind of vehicle.
2. I think Vehicle too is not the right name as it implies transportation of goods or people. Here we are dealing with robots with wheels.
4. "Create Vehicle" should be a button in the CarComponent, like "Build Arm" is for RobotArms.
3. CarComponent should extend RobotComponent, e.g. Inherit the "Go to home function" (https://github.com/MarginallyClever/Robot-Overlord-App/pull/157)
4. While I haven't yet looked at the code, I think it's wrong for a CarComponent to know the types of the wheel.
A wheel should know what type it is.
What about mixed configuration?
6. Deleting a suspension freeze the renderer an flood the console with Null Pointer Exception.
I think it should be better handled.
7. 2.7.0 is still unusable on MacOS (https://github.com/MarginallyClever/Robot-Overlord-App/issues/151)

i-make-robots, 2023-06-13:
we debated it for about an hour.  I couldn't find a better word than "Vehicle".

Jackymancs4, 2023-06-13:
> we debated it for about an hour. I couldn't find a better word than "Vehicle".

Who and when?
Anyway I vote for the original idea of WheeledRobotComponent (like ArmRobotComponent).
Google seems to appreciate it as it finds many results.
Used also in https://en.wikibooks.org/wiki/Robotics/Types_of_Robots/Wheeled and https://en.wikipedia.org/wiki/Differential_wheeled_robot

1. At least I suggest to replace CarComponent with VehicleComponent
2. agree
3. I'm gonna open a separate issue for this as I have ideas and considerations on it.
4. I think they should. If I drive my vehicle I have no obvious way to undo it, as in reposition on orgin with throttle and steer perfectly at 0. I'll update my PR with a proposal.
5. I'll think about it later.
6. Will open a new issue with exception.

i-make-robots, 2023-06-13:
IMHO Throttle and Steer to zero should be in the drive panel.  Returning in to a given location is transform and rotate tools.  In a larger system where there are simulations of many vehicles (say, in a warehouse simulation) "home" loses all meaning.  It feels like we're trying to use the robot arm "home" for some kind of reset feature, which is doing two different things with the same button.

Jackymancs4, 2023-06-13:
> Throttle and Steer to zero should be in the drive panel. 

This deserve a proper issue. But yes, the sliders are a little hard to get right.

>  In a larger system where there are simulations of many vehicles (say, in a warehouse simulation) "home" loses all meaning.

I don't see how. If I have twenty car robot positioned in a specific layout, then I drive all of them around, I need a way to getting all of them to return at home, with home as in "Initial position", not the origin. Surely the transform and rotate tools are not right for the job. I'm not saying to the robot "Drive to home" (for that I would use a targetComponent), I'm saying just get back anew.

> the robot arm "home" for some kind of reset feature

That is surely the way I use it in the ArmRobot (most likely in all other kind of robots like Dogs) and what i would need for the Vehicles. I guess especially if I have a fleet.

> which is doing two different things with the same button.

Two things?

closed 2023-08-21 by github-actions[bot]