MarginallyClever/Robot-Overlord-App #217: SmallTalk interpreter in Java > drive robot

i-make-robots, 2024-01-04:
**Is your feature request related to a problem? Please describe.**
Jon would like to drive his robots using SmallTalk "as though it was in a ROS node".
This implies that nodes should talk to each other with the ROS system.

JonHylands, 2024-01-04:
I don't have enough experience with ROS to say whether or not this is true... The current controller of my robots (running on a Raspberry pi) is written in Squeak Smalltalk, and does not have anything to do with ROS.

i-make-robots, 2024-01-05:
@JonHylands awesome.  Have you got Smalltalk samples of your current work?  I'm interested in the API you use to drive motors and read sensors.

JonHylands, 2024-01-05:
For Roz and MicroRaptor, it is very simple - everything other than their respective cameras are on a serial bus. The servos for each robot are Robotis Dynamixel servos, AX-12s to be specific. Each servo has its own unique id (1-252), and they are all daisy chained on a serial bus. Also on this serial bus are a couple STM32F405 boards, which have specific sensors attached (IMU, Optical flow, ToF LIDAR, etc). The 405 boards run MicroPython, and I've written code that allows the 405 to be presented on the bus as just another serial device. Each device on the bus has a control table, and in the case of the 405's, the control table contain sensor values that are updated at some refresh rate (might be 50Hz or 100Hz, I don't remember off the top of my head).

The bus is presented to the Raspberry pi, and thus to Smalltalk, as a serial port, with the comms speed set to one megabit. My code in Smalltalk can query the bus, find out what devices are connected, and talk to each device individually, all over this single serial port.

i-make-robots, 2024-01-09:
Great... what does the API look like?  At my level all I care about is how to connect your API with mine.

JonHylands, 2024-01-10:
Are you talking about the API for the serial bus, or the robot? The robot doesn't have an API. The serial bus API is well documented by Robotis (https://emanual.robotis.com/docs/en/dxl/ax/ax-12a/) and (https://emanual.robotis.com/docs/en/dxl/protocol1/). My new devices follow the same structure and protocol, although the control table is obviously different.