MarginallyClever/Robot-Overlord-App #281: Endless Exceptions in log

makiam, 2024-03-27:
**Describe the bug**
I see endless Exceptions in log

**To Reproduce**
Steps to reproduce the behavior:
Clone repository in Idea 
Run RO3.java class
Exception enldessly throws in IDE  log: (see at bottom...)

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
 - Device: PC with Intel UHD graphics 630
 - OS: Ubuntu 20.04 lTS
 - App Version [e.g. 22]
 - Java version: Open JDK 20

**Additional context**
Add any other context about the problem here.

Stacktrace:
java.lang.reflect.InvocationTargetException
	at java.desktop/java.awt.EventQueue.invokeAndWait(EventQueue.java:1371)
	at java.desktop/java.awt.EventQueue.invokeAndWait(EventQueue.java:1346)
	at java.desktop/javax.swing.SwingUtilities.invokeAndWait(SwingUtilities.java:1480)
	at jogl.all@2.5.0/com.jogamp.opengl.util.AWTAnimatorImpl.display(AWTAnimatorImpl.java:98)
	at jogl.all@2.5.0/com.jogamp.opengl.util.AnimatorBase.display(AnimatorBase.java:471)
	at jogl.all@2.5.0/com.jogamp.opengl.util.FPSAnimator$MainTask.run(FPSAnimator.java:245)
	at java.base/java.util.TimerThread.mainLoop(Timer.java:566)
	at java.base/java.util.TimerThread.run(Timer.java:516)
Caused by: com.jogamp.opengl.GLException: Caught NullPointerException: Cannot invoke "com.marginallyclever.ro3.apps.viewport.ShaderProgram.use(com.jogamp.opengl.GL3)" because "this.meshShader" is null on thread AWT-EventQueue-0
	at jogl.all@2.5.0/com.jogamp.opengl.GLException.newGLException(GLException.java:76)
	at jogl.all@2.5.0/jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1327)
	at jogl.all@2.5.0/jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at jogl.all@2.5.0/com.jogamp.opengl.awt.GLJPanel$OffscreenBackend.doPaintComponent(GLJPanel.java:2127)
	at jogl.all@2.5.0/com.jogamp.opengl.awt.GLJPanel.paintComponent(GLJPanel.java:587)
	at java.desktop/javax.swing.JComponent.paint(JComponent.java:1128)
	at java.desktop/javax.swing.JComponent.paintToOffscreen(JComponent.java:5318)
	at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBufferedImpl(RepaintManager.java:1656)
	at java.desktop/javax.swing.RepaintManager$PaintManager.paintDoubleBuffered(RepaintManager.java:1631)
	at java.desktop/javax.swing.RepaintManager$PaintManager.paint(RepaintManager.java:1569)
	at java.desktop/javax.swing.RepaintManager.paint(RepaintManager.java:1336)
	at java.desktop/javax.swing.JComponent._paintImmediately(JComponent.java:5266)
	at java.desktop/javax.swing.JComponent.paintImmediately(JComponent.java:5076)
	at java.desktop/javax.swing.RepaintManager$4.run(RepaintManager.java:878)
	at java.desktop/javax.swing.RepaintManager$4.run(RepaintManager.java:861)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:400)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:87)
	at java.desktop/javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:861)
	at java.desktop/javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:834)
	at jogl.all@2.5.0/com.jogamp.opengl.util.AWTAnimatorImpl$1.run(AWTAnimatorImpl.java:170)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:308)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:720)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:714)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:400)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:87)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
Caused by: java.lang.NullPointerException: Cannot invoke "com.marginallyclever.ro3.apps.viewport.ShaderProgram.use(com.jogamp.opengl.GL3)" because "this.meshShader" is null
	at com.marginallyclever.robotoverlord/com.marginallyclever.ro3.apps.viewport.renderpasses.DrawMeshes.drawAllMeshes(DrawMeshes.java:286)
	at com.marginallyclever.robotoverlord/com.marginallyclever.ro3.apps.viewport.renderpasses.DrawMeshes.draw(DrawMeshes.java:189)
	at com.marginallyclever.robotoverlord/com.marginallyclever.ro3.apps.viewport.Viewport.renderAllPasses(Viewport.java:503)
	at com.marginallyclever.robotoverlord/com.marginallyclever.ro3.apps.viewport.Viewport.display(Viewport.java:467)
	at jogl.all@2.5.0/jogamp.opengl.GLDrawableHelper.displayImpl(GLDrawableHelper.java:692)
	at jogl.all@2.5.0/jogamp.opengl.GLDrawableHelper.display(GLDrawableHelper.java:674)
	at jogl.all@2.5.0/com.jogamp.opengl.awt.GLJPanel$Updater.display(GLJPanel.java:1478)
	at jogl.all@2.5.0/com.jogamp.opengl.awt.GLJPanel$9.run(GLJPanel.java:1553)
	at jogl.all@2.5.0/jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
	... 31 more

i-make-robots, 2024-03-27:
Thank you for your report!  I'm running it right now and I'm not getting the same endlessly repeating error.  Is there anything at the start of the log about the meshShader failing to create?

i-make-robots, 2024-03-27:
src/main/java/com/marginallyclever/ro3/apps/viewport/renderpasses/DrawMeshes.java line 72.
"Failed to load mesh shader" is the error if it doesn't load, with more info after that.

makiam, 2024-03-28:
Hi.
This is more log from the application start.
14:58:40.150 INFO  ------------------------------------------------
14:58:40.537 INFO  availability=Natives[GL4bc true [4.6 (Compat profile, arb, compat[ES2, ES3, ES31], FBO, hardware)], GL4 true [4.6 (Core profile, arb, compat[ES2, ES3, ES31, ES32], FBO, hardware)], GLES3 true [3.2 (ES profile, arb, compat[ES2, ES3, ES31, ES32], FBO, hardware)], GL3bc true [4.6 (Compat profile, arb, compat[ES2, ES3, ES31], FBO, hardware)], GL3 true [4.6 (Core profile, arb, compat[ES2, ES3, ES31, ES32], FBO, hardware)], GL2 true [4.6 (Compat profile, arb, compat[ES2, ES3, ES31], FBO, hardware)], GLES2 true [3.2 (ES profile, arb, compat[ES2, ES3, ES31, ES32], FBO, hardware)], GLES1 true [1.1 (ES profile, arb, compat[FP32], hardware)], count 8 / 8], Common[, GL4ES3 true, GL2GL3 true, GL2ES2 true, GL2ES1 true], Mappings[GL3bc GLProfile[GL3bc/GL4bc.hw], GL2 GLProfile[GL2/GL4bc.hw], GLES3 GLProfile[GLES3/GLES3.hw], GL4 GLProfile[GL4/GL4.hw], GL3 GLProfile[GL3/GL4.hw], GL2GL3 GLProfile[GL2GL3/GL4bc.hw], GLES1 GLProfile[GLES1/GLES1.hw], GLES2 GLProfile[GLES2/GLES3.hw], GL2ES1 GLProfile[GL2ES1/GL4bc.hw], GL4ES3 GLProfile[GL4ES3/GL4.hw], GL2ES2 GLProfile[GL2ES2/GL4.hw], GL4bc GLProfile[GL4bc/GL4bc.hw], , default GLProfile[GL4bc/GL4bc.hw], count 12 / 12]
14:58:40.538 INFO  capabilities=rgba 8/8/8/1, opaque, accum-rgba 0/0/0/0, dp/st/ms 16/8/4, sample-ext default, dbl, mono  , hw, GLProfile[GL4bc/GL4bc.hw], on-scr[.]
14:58:40.538 INFO  create canvas
hourAngle=12.0 declination=0.0
14:58:40.647 DEBUG Found 29 matches
14:58:40.648 DEBUG Grown to 29 matches
14:58:40.677 INFO  Resetting layout to default.
14:58:40.694 INFO  done.
14:58:40.694 DEBUG /home/huawei/RobotOverlord/ro3.layout
14:58:40.803 DEBUG adding drag + drop support...
14:58:40.822 DEBUG load RenderPass State
14:58:40.920 INFO  init
14:58:40.926 ERROR Failed to compile fragment shader code: 0:44(25): error: `return' with wrong type int, in function `ShadowCalculation' returning float
 
14:58:40.941 ERROR Failed to load mesh shader
java.lang.IllegalStateException: Failed to link shader program.
	at com.marginallyclever.robotoverlord/com.marginallyclever.ro3.apps.viewport.ShaderProgram.<init>(ShaderProgram.java:41)
	at com.marginallyclever.robotoverlord/com.marginallyclever.ro3.apps.viewport.renderpasses.DrawMeshes.init(DrawMeshes.java:70)
	at jogl.all@2.5.0/jogamp.opengl.GLDrawableHelper.reshape(GLDrawableHelper.java:749)
	at jogl.all@2.5.0/com.jogamp.opengl.awt.GLJPanel$Updater.display(GLJPanel.java:1474)
	at jogl.all@2.5.0/com.jogamp.opengl.awt.GLJPanel$9.run(GLJPanel.java:1553)
	at jogl.all@2.5.0/jogamp.opengl.GLDrawableHelper.invokeGLImpl(GLDrawableHelper.java:1293)
	at jogl.all@2.5.0/jogamp.opengl.GLDrawableHelper.invokeGL(GLDrawableHelper.java:1147)
	at jogl.all@2.5.0/com.jogamp.opengl.awt.GLJPanel$OffscreenBackend.doPaintComponent(GLJPanel.java:2127)
	at jogl.all@2.5.0/com.jogamp.opengl.awt.GLJPanel.paintComponent(GLJPanel.java:587)
	at java.desktop/javax.swing.JComponent.paint(JComponent.java:1128)
	at java.desktop/javax.swing.JComponent.paintChildren(JComponent.java:961)
	at java.desktop/javax.swing.JComponent.paint(JComponent.java:1137)
	at java.desktop/javax.swing.JComponent.paintChildren(JComponent.java:961)
	at java.desktop/javax.swing.JComponent.paint(JComponent.java:1137)
	at java.desktop/javax.swing.JComponent.paintChildren(JComponent.java:961)
	at java.desktop/javax.swing.JComponent.paint(JComponent.java:1137)
	at java.desktop/javax.swing.JComponent.paintChildren(JComponent.java:961)
	at java.desktop/javax.swing.JComponent.paint(JComponent.java:1137)
	at java.desktop/javax.swing.JComponent.paintChildren(JComponent.java:961)
	at java.desktop/javax.swing.JComponent.paint(JComponent.java:1137)
	at java.desktop/javax.swing.JComponent.paintChildren(JComponent.java:961)
	at java.desktop/javax.swing.JSplitPane.paintChildren(JSplitPane.java:1034)
	at java.desktop/javax.swing.JComponent.paint(JComponent.java:1137)
	at java.desktop/javax.swing.JComponent.paintToOffscreen(JComponent.java:5318)
	at java.desktop/javax.swing.BufferStrategyPaintManager.paint(BufferStrategyPaintManager.java:246)
	at java.desktop/javax.swing.RepaintManager.paint(RepaintManager.java:1336)
	at java.desktop/javax.swing.JComponent._paintImmediately(JComponent.java:5266)
	at java.desktop/javax.swing.JComponent.paintImmediately(JComponent.java:5076)
	at java.desktop/javax.swing.RepaintManager$4.run(RepaintManager.java:878)
	at java.desktop/javax.swing.RepaintManager$4.run(RepaintManager.java:861)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:400)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:87)
	at java.desktop/javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:861)
	at java.desktop/javax.swing.RepaintManager.paintDirtyRegions(RepaintManager.java:834)
	at jogl.all@2.5.0/com.jogamp.opengl.util.AWTAnimatorImpl$1.run(AWTAnimatorImpl.java:170)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:308)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:720)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:714)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:400)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:87)
[... cut ...]
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
14:58:40.945 DEBUG Creating shadow FBO
14:58:40.945 DEBUG Creating depth map
14:58:40.946 DEBUG Binding depth map 2 to shadow FBO

i-make-robots, 2024-03-28:
```14:58:40.926 ERROR Failed to compile fragment shader code: 0:44(25): error: return' with wrong type int, in function ShadowCalculation' returning float

14:58:40.941 ERROR Failed to load mesh shader
java.lang.IllegalStateException: Failed to link shader program.
at com.marginallyclever.robotoverlord/com.marginallyclever.ro3.apps.viewport.ShaderProgram.(ShaderProgram.java:41)
at com.marginallyclever.robotoverlord/com.marginallyclever.ro3.apps.viewport.renderpasses.DrawMeshes.init(DrawMeshes.java:70)```

Aha.

i-make-robots, 2024-03-28:
That's @#$*( stupid.  I return 0, which is zero in every type.  Your driver is very pedantic.  I've changed the code.  Please sync and try again.

makiam, 2024-03-29:
No more exceptions and UI renders OK.
Thanks

i-make-robots, 2024-03-29:
Great!  Join us on discord and share what you're doing with the app.  I'm here to help!  https://discord.gg/jcCebWZ3

closed 2024-03-29 by i-make-robots